basler/pylon-ros-camera #264: can't capture image at defined frequency.

leiwanrobotics, 2025-02-03:
### Issue description

I have a basler camera, a2A1920-51gcBAS

I tried this driver and can reach the defined frequency under free mode(trigger mode = off). But it will heavly increase the usage of bandwidth (easily reach 850 Mbps). I want to decrease the usage of bandwidth as I have multi cameras, so I decide to use trigger mode on. However, the camera can't reach the defined frequency anymore.

When trigger mode = off, I also got error:

[image]

Do you have any solution?

BTY, how can I capture image with 20 / 30 Hz starting from interger seconds, like 0s, 1s and so on.

### Code example about how to reproduce the issue

ROS2 driver

### Error message

```shell
[ERROR] [1738586959.635602687] [basler.pylon.ros2.pylon_ros2_camera_base]: An image grabbing exception occurred: Grab timed out. Possible reasons are: The image transport from the camera device is not working properly, e.g., all GigE network packets for streaming are dropped; The camera uses explicit triggering (see TriggerSelector for more information) and has not been triggered; Single frame acquisition mode is used and one frame has already been acquired; The acquisition has not been started or has been stopped.
[ERROR] [1738586959.635730926] [basler.pylon.ros2.pylon_ros2_camera_base]: Error: Grab was not successful
```

### Hardware setup description

- Ubuntu22.04
- ROS2 Humble

### Runtime information

```shell
[WARN] [1738586109.830685923] [basler.pylon.ros2.pylon_ros2_camera_parameter]: Gain and exposure are specified as startup parameters and hence assumed to be fixed! The specified brightness (100) can't be reached! Brightness is going to ignored by only setting gain and exposure.
[INFO] [1738586109.830757795] [basler.pylon.ros2.pylon_ros2_camera_parameter]: Autoflash: 0, line2: 1, line3: 1
[INFO] [1738586109.830790854] [basler.pylon.ros2.pylon_ros2_camera_parameter]: Trying to connect the camera device with the following device user id: 40549247
[INFO] [1738586111.619580333] [basler.pylon.ros2.pylon_ros2_camera]: Found camera device! Device Model: a2A1920-51gcBAS with Device User Id: 40549247
[INFO] [1738586111.946480993] [basler.pylon.ros2.pylon_ros2_camera_node]: Pylon camera instance created with the following user id: 40549247
[INFO] [1738586112.143901373] [basler.pylon.ros2.pylon_ros2_gige_ace2_camera]: Startup user profile set to "UserSet3"
[INFO] [1738586112.210065024] [basler.pylon.ros2.pylon_ros2_gige_ace2_camera]: UserSet3 loaded
[INFO] [1738586112.477361361] [basler.pylon.ros2.pylon_ros2_camera_node]: CameraInfoURL needed for rectification! ROS2-Param: '//camera_info_url' = '' is invalid!
[DEBUG] [1738586112.477408762] [basler.pylon.ros2.pylon_ros2_camera_node]: CameraInfoURL should have following style: 'file:///full/path/to/local/file.yaml' or 'package://<package_name>/relative/path/to/file.yaml'
[WARN] [1738586112.477418594] [basler.pylon.ros2.pylon_ros2_camera_node]: Will only provide distorted /image_raw images!
[WARN] [1738586112.477430013] [basler.pylon.ros2.pylon_ros2_camera_base]: Camera does not support binning (X). Will keep the current settings.
[INFO] [1738586112.477544708] [basler.pylon.ros2.pylon_ros2_camera_node]: Setting horizontal binning_x to 1
[WARN] [1738586112.477557656] [basler.pylon.ros2.pylon_ros2_camera_base]: Camera does not support binning (Y). Will keep the current settings.
[INFO] [1738586112.477635291] [basler.pylon.ros2.pylon_ros2_camera_node]: Setting vertical binning_y to 1
[INFO] [1738586112.486854453] [basler.pylon.ros2.pylon_ros2_camera_node]: Attempted to set exposure to 10000, reached: 10000
[INFO] [1738586112.492703423] [basler.pylon.ros2.pylon_ros2_camera_node]: Attempted to set gain to: 0, reached: 0
[INFO] [1738586112.504546509] [basler.pylon.ros2.pylon_ros2_camera_node]: Attempted to set gamma to 1, reached: 1
[INFO] [1738586112.509418444] [basler.pylon.ros2.pylon_ros2_camera_node]: Startup settings: encoding = 'bayer_rggb8', binning = [1, 1], exposure = 10000, gain = 0, gamma = 1, shutter mode = global
[INFO] [1738586112.512961981] [basler.pylon.ros2.pylon_ros2_camera_node]: Desired framerate 30.00 is higher than max possible. Will limit framerate to: 30.00 Hz
[INFO] [1738586112.514502974] [basler.pylon.ros2.pylon_ros2_camera_node]: Start image grabbing if node connects to topic with a spinning rate of: 29.9994 Hz
[INFO] [1738586112.548128207] [basler.pylon.ros2.pylon_ros2_camera_node]: Camera not calibrated
[ERROR] [1738586959.635602687] [basler.pylon.ros2.pylon_ros2_camera_base]: An image grabbing exception occurred: Grab timed out. Possible reasons are: The image transport from the camera device is not working properly, e.g., all GigE network packets for streaming are dropped; The camera uses explicit triggering (see TriggerSelector for more information) and has not been triggered; Single frame acquisition mode is used and one frame has already been acquired; The acquisition has not been started or has been stopped.
[ERROR] [1738586959.635730926] [basler.pylon.ros2.pylon_ros2_camera_base]: Error: Grab was not successful
```

### Is your camera operational with the Basler pylon Viewer on your platform?

Yes

DanBmh, 2025-02-25:
> BTY, how can I capture image with 20 / 30 Hz starting from interger seconds, like 0s, 1s and so on.

You might be interesed in: https://github.com/basler/pylon-ros-camera/pull/263

leiwanrobotics, 2025-04-28:
@FrancoisPicardDTI 
Thanks for your reply. I have two problem.
1. Actually I have three sensors in the system, LiDAR, Thermal camera and RGB camera basler. As suggested by otehr issue report, I have to set Acquisition Mode = Continuous + Trigger Mode = Off, to get the expected frame rate. However, I saw this setup will occupy lots of network bandwidth because of continuous mode. When I launch all sensors, then RGB can't actually get expected frame rate and decrease from 30 to 20Hz. 
2. I want to not just reach the expected frequency but also control the capturing time of image. The capture time of camera should be even, such as 0.0, 0.1, ....1.0s, their time interval should be always around 0.1s. But actually the data recorded shows the time interval between frames varies from 0.05 - 0.3s.

leiwanrobotics, 2025-04-28:
@DanBmh 
Thanks for your reply.

May I ask how did you set the Acquisition Mode of camera? Software trigger or Continuous ? How much frequency can you reach? Is it still possible to reach 30 Hz?

DanBmh, 2025-04-29:
> May I ask how did you set the Acquisition Mode of camera? Software trigger or Continuous ? How much frequency can you reach? Is it still possible to reach 30 Hz?

Software trigger. Similar as in the other issues about it, it is slower than the maximum camera frame rate (about half of it). My addition doesn't add a notable delay.

FrancoisPicardDTI, 2026-10-29:
No more activity here, I am closing the issue for now. Reopen it if needed.

closed 2026-10-29 by FrancoisPicardDTI