basler/pylon-ros-camera #265: Continuous Brightness, Auto Exposure, and Auto Gain Not Working in Basler ace 2 R (a2A1920-160ucBAS)

MiniiParra, 2025-02-12:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I am currently working with the Basler ace 2 R (a2A1920-160ucBAS) camera and have successfully set up the pylon SDK following the instructions from the GitHub repository, using the humble branch. However, I am facing an issue with the configuration parameters for continuous brightness, auto exposure, and auto gain control.

I have set the following parameters in the configuration file:

brightness_continuous: true

exposure_auto: true

gain_auto: true

Despite these settings, the camera is not adjusting to different lighting conditions or environments. The image brightness, exposure, and gain do not adapt as expected.

**Steps to reproduce:**

Set the configuration parameters in the YAML file as mentioned above.
Run the launch camera with the pylon driver in the ROS environment.
Observe that the camera does not adapt to changes in lighting or exposure.

### Hardware setup description

Camera: Basler ace 2 R (a2A1920-160ucBAS)
PC/Embedded system: Jetson Orin AGX
Operating System: Ubuntu 22
RAM: 64G
CPU architecture: ARM64
Interfaces used: USB

### Runtime information

```shell
I have the full Basler pylon Camera Software Suite installed.
ROS version: ROS2 Humble
Camera works in the Basler pylon Viewer on the platform, and the settings there seem to be functional.
```

### Is your camera operational with the Basler pylon Viewer on your platform?

Yes

FrancoisPicardDTI, 2025-02-20:
Hello @MiniiParra 

You need to set a value for brightness as well. This is the brightness value that set the pace for the adjusting of the exposure time and the gain.

**Documentation for the parameter 'brightness_continuous':**

> Only relevant, if 'brightness' is set.
> The brightness_continuous flag controls the auto brightness function. If it is set to false, the brightness will only be reached once. Hence changing light conditions lead to changing brightness values. If it is set to true, the given brightness will be reached continuously, trying to adapt to changing light conditions. This is only possible for values in the possible auto range of the pylon API which is generally [50 - 205].

**Documentation for the parameters 'exposure_auto' and 'gain_auto'**

> Only relevant, if 'brightness' is set.
> If the camera should try to reach and / or keep the brightness, hence adapting to changing light conditions, at least one of the following flags must be set. If both are set, the interface will use the profile that tries to keep the gain at minimum to reduce white noise. The exposure_auto flag indicates, that the desired brightness will be reached by adapting the exposure time. The gain_auto flag indicates, that the desired brightness will be reached by adapting the gain.

jack-buffington, 2025-02-20:
I'm also having this issue.  I've tried every permutation I can think of for the settings.  I have two a2A1920-51gcPRO cameras.  Both are acting in the same manner.  
I have tried this on both ARM64 and X86_64 processors.

What I am seeing when I try to do autoexposure is that the camera's image adjusts but goes to pure white.  In other words, it appears to be continuously increasing the exposure time.     If I try to relaunch my ros nodes, the settings seem to have stayed on the camera and I can't connect without cycling the power to the camera.  

Example settings that cause this:
exposure: 1000  (This appears to be what the camera will start at)
brightness: 100 (I'm assuming a range of 0->255
brightness_continuous: true
exposure_auto: true

FrancoisPicardDTI, 2026-10-29:
Closing the issue as it seems the fix solved the issue

closed 2026-10-29 by FrancoisPicardDTI