basler/pylon-ros-camera #266: Ros2 Foxy versión

XabierSener, 2025-02-12:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Good afternoon,

Thank you for providing the ROS driver for the Basler camera. I'm reaching out because I need to record the state of a road for a project. I have an NVIDIA Jetson AGX Xavier that I want to connect the Basler camera to. Additionally, I have other devices that only work with ROS2 Foxy. I would like to know if you have implemented drivers for this version. I would greatly appreciate your assistance.

Thank you very much.
Xabier

### Hardware setup description

Basler Ace 2 Ra2A2840-67g5cBAS. (Camera).  Nvidia jetson Agx Xavier (onboard computer)

### Runtime information

```shell
Ros 2 foxy
```

### Is your camera operational with the Basler pylon Viewer on your platform?

Yes

FrancoisPicardDTI, 2025-02-18:
Hello @XabierSener 
The driver has not been tested under Foxy. But did you try the [galactic](https://github.com/basler/pylon-ros-camera/tree/galactic) branch? It should not be far from something working under Foxy.
Let me know if you try and what you experience.

FrancoisPicardDTI, 2025-06-03:
I am closing this one as it seems that it is working according to #276

closed 2025-06-03 by FrancoisPicardDTI