basler/pylon-ros-camera #268: BUG: Cannot run drivers in parallel on Jazzy

ndahn, 2025-03-06:
### Issue description

We have a setup with multiple acA2040-25ac cameras. Since upgrading to ROS2 Jazzy, we found that it is no longer possible to run multiple camera drivers in parallel.

Starting a driver instance (component) for either of our cameras works just fine. However, it is not possible to run a second driver instance. It either crashes before doing anything useful, sometimes even before initializing, or it kills the already running instance. 

### Code example about how to reproduce the issue

This is our launch file. I have also tested with two separate launch files. 

[start_cameras.launch.py.txt](https://github.com/user-attachments/files/19114122/start_cameras.launch.py.txt)

### Error message

```shell
Attached are logs from an especially egregious instance where our left camera was killed by the right camera, then killed the right camera while restarting. This repeated two more times until finally the right camera driver crashed.

[basler_log_left.txt](https://github.com/user-attachments/files/19113955/basler_log_left.txt)
[basler_log_right.txt](https://github.com/user-attachments/files/19113954/basler_log_right.txt)

The following log was created using the launch file mentioned below, starting both cameras as part of a single launch file.

[basler_log_both.txt](https://github.com/user-attachments/files/19114118/basler_log_both.txt)
```

### Hardware setup description

Both cameras are attached to a switch which is connected to an Nvidia Jetson Orin (currently via a 1gb network cable). The camera drivers are started in a docker container with ROS2 Jazzy. Grabbing frames from a single camera driver works. Grabbing frames from two cameras using the `pylonviewer` works (as long as the framerate is low enough).

### Runtime information

```shell
Output captured from the Jetson host system (not the docker container).

pylon info: 9.0.3.215 64-Bit
----------------------------------------
pylon Setup info: 8.0.0.16021 
----------------------------------------
pylon Application info: 3.3.0.16064 64-Bit
----------------------------------------
pylon Viewer info: 8.1.1.16064 64-Bit
----------------------------------------
Host system infos:

CPU architecture: arm64
Kernel type: linux
Kernel version: 5.15.136-tegra
OS type: ubuntu
OS version: 22.04
Render Backend: OpenGL
----------------------------------------
Screen(s):

Screen 0 scale factor: 1
Screen 0 resolution: 2560x1440
Screen 1 scale factor: 1
Screen 1 resolution: 1920x1080
----------------------------------------
Window(s):

"pylon Viewer 64-Bit" on screen 0
----------------------------------------
Host network infos:

Host network adapter mgbe0:
    Name: mgbe0
    MAC address: 48:B0:2D:EF:0F:C0
    IP address: 192.168.0.2
    Subnet mask: 255.255.0.0

Host network adapter mgbe1:
    Name: mgbe1
    MAC address: 48:B0:2D:EF:0F:C1
    IP address: 192.168.0.253
    Subnet mask: 255.255.255.255
    IP address: 192.168.0.253
    Subnet mask: 255.255.0.0

Host network adapter docker0:
    Name: docker0
    MAC address: 02:42:F1:47:4D:8F
    IP address: 172.17.0.1
    Subnet mask: 255.255.0.0

Host network adapter l4tbr0:
    Name: l4tbr0
    MAC address: 3E:C7:E7:79:F3:87

Host network adapter br-fe68320e54fe:
    Name: br-fe68320e54fe
    MAC address: 02:42:8F:F9:13:14
    IP address: 172.18.0.1
    Subnet mask: 255.255.0.0

Host network adapter usb0:
    Name: usb0
    MAC address: D6:42:7C:34:4B:09

Host network adapter usb1:
    Name: usb1
    MAC address: D6:42:7C:34:4B:0B

Host network adapter veth18a2174:
    Name: veth18a2174
    MAC address: EA:1F:77:25:DA:C1
----------------------------------------
Device infos:

GigE Vision device:
    Model name: acA2040-25gc
    Manufacturer name: Basler
    Serial number: 22141561
    MAC address: 00:30:53:20:AD:79
    IP address: 192.168.1.1
    Subnet mask: 255.255.0.0

GigE Vision device:
    Model name: acA2040-25gc
    Manufacturer name: Basler
    Serial number: 22141559
    MAC address: 00:30:53:20:AD:77
    IP address: 192.168.1.2
    Subnet mask: 255.255.0.0

GigE Vision device:
    Model name: acA2040-25gc
    Manufacturer name: Basler
    Serial number: 22141557
    MAC address: 00:30:53:20:AD:75
    IP address: 192.168.1.3
    Subnet mask: 255.255.0.0

GigE Vision device:
    Model name: acA2040-25gc
    Manufacturer name: Basler
    Serial number: 22141557
    MAC address: 00:30:53:20:AD:75
    IP address: 192.168.1.3
    Subnet mask: 255.255.0.0

GigE Vision device:
    Model name: acA2040-25gc
    Manufacturer name: Basler
    Serial number: 22141561
    MAC address: 00:30:53:20:AD:79
    IP address: 192.168.1.1
    Subnet mask: 255.255.0.0

GigE Vision device:
    Model name: acA2040-25gc
    Manufacturer name: Basler
    Serial number: 22141559
    MAC address: 00:30:53:20:AD:77
    IP address: 192.168.1.2
    Subnet mask: 255.255.0.0
```

### Is your camera operational with the Basler pylon Viewer on your platform?

Yes

FrancoisPicardDTI, 2025-03-18:
Hello @ndahn 

I don't manage to reproduce your issue. I tested with 2 cameras, and both with 2 launch files and one launch file starting 2 nodes. For the latter experiment, my launch file is different though. Maybe you can try this approach.

In your logs, I can see a couple of memory errors when reading parameters ("An exception while getting the camera current params occurred:Failed to read memory at 0x20024, 0x4 bytes. Timeout, no message received. (0xE101800B)"). Maybe it leads afterwards to your issue.

There's as well an image grabbing exception at the end: "Possible reasons are: The image transport from the camera device is not working properly, e.g., all GigE network packets for streaming are dropped; The camera uses explicit triggering (see TriggerSelector for more information) and has not been triggered; Single frame acquisition mode is used and one frame has already been acquired; The acquisition has not been started or has been stopped."
Did you investigate these points?

FrancoisPicardDTI, 2025-03-18:
[two_cams.launch.py.txt](https://github.com/user-attachments/files/19316798/two_cams.launch.py.txt)

ndahn, 2025-03-18:
Thank you @F@FrancoisPicardDTI for getting back to me! This issue appears on our side when using composition; for performance reasons, using normal nodes is unfortunately not an option for us. Did you try it using composition as well?

The first error you noticed happens when a second driver instance starts and crashes the other one. In the `basler_log_both.txt` I provided it happens several times as the two driver instances crash each other. 

The second error occurs when a frame is not completely grabbed, e.g. due to bandwidth restrictions. In our logs this only happened at the very end when I terminated the process, so I don't think it's related here.

FrancoisPicardDTI, 2025-03-18:
I won't be able to test it today. But I'll come back to you as soon as I can.

ndahn, 2025-03-18:
I investigated further using the launch file you provided, which resulted in the same issue. It seems like using any of the UserSets (1, 2 or 3) leads to a crash, while *Default* and *CurrentSettings* are fine. I don't think we have ever used UserSet3, but I will reset our camera settings tomorrow just to be sure.

ndahn, 2025-04-14:
We reset all camera parameters and it seems like that solved our issue. Unfortunately I have no idea which settings could have caused these problems - whatever it was, it's not something we rely on. Will close this for now - thank you for your support Francois!

closed 2025-04-14 by ndahn

FrancoisPicardDTI, 2025-04-15:
Thank you @ndahn !