basler/pylon-ros-camera #272: Rosdep support for pylon installation

J-C-Haley, 2025-04-17:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Hello and thank you in advance,

Are there plans to host a rosdep-installable pylon version? If not, can the reason for this be added to the Readme documentation? Currently it states:

"From [pylon Camera Software Suite](https://www2.baslerweb.com/en/downloads/software-downloads/) version 7.5.0 or newer. The latest APi libraries must be installed manually."

This was not the case for the old ROS1 noetic install, which could download and install version 6.2.0 for you. This was a major convenience for automating installation, docker containers, and CI of apps that depended on pylon. For now the only alternative seems to be to force users to the basler website?

I would be highly interested in any alternatives to rosdep you could recommend that would make scalable use of your great software easier. I'm also unclear as to if the API license disallows redistribution, in case we were to bundle a copy with our software (for instance, in a container)?

Thank you very much for the clarification! Wish to respect the usage wishes, I hope we can find a solution.

### Hardware setup description

Ace classic, acA5472-17um
X86, Ubuntu 24.04, 64GB RAM
USB primarily

### Runtime information

```shell
Pylon 8.0.0
ROS2 Jazzy
Installed in a docker container:
BASE=nvidia/cuda:12.6.1-base-ubuntu24.04
```

### Is your camera operational with the Basler pylon Viewer on your platform?

Yes

FrancoisPicardDTI, 2025-04-23:
Hello @J-C-Haley 
For now, there is no plan for rosdep support for the pylon API installation. Sorry about that.
About the redistribution, it is allowed, yes, provided that someconditions are met. See here: https://docs.baslerweb.com/licensing-information.

FrancoisPicardDTI, 2025-04-29:
Hello @J-C-Haley 
To complete my post related to the API redistribution, the pylon Camera Software Suite, including the API and SDK, is subject to Basler's proprietary license terms. According to the current End User License Agreement (EULA), redistribution of the pylon libraries, for example by bundling them with your own software or including them in a Docker container, is permitted under the following conditions:
•	The software must be used in connection with Basler hardware (e.g., Basler cameras).
•	The pylon libraries must remain unmodified.
•	The Basler license terms must be passed on to the end users (e.g., by including the EULA or linking to it).
•	The pylon software must not be marketed or distributed independently from Basler products.

This means that you can bundle the pylon API with your application, including within a containerized environment, provided it is intended for use with Basler hardware. Please have a look at the latest version of the pylon EULA.
If you require a formal confirmation for commercial redistribution scenarios, you can contact the Basler Support directly at [support@baslerweb.com](mailto:support@baslerweb.com).

J-C-Haley, 2025-05-15:
Great, this should work for us, thank you very much for confirming the terms. Current plan is to just use git submodules to host the pylon download and pass along the EULA and above conditions. May revisit the idea and make it into a ros package (that should be rosdep compatible, and would come independent of your driver). Thanks!

closed 2025-05-15 by J-C-Haley