basler/pylon-ros-camera #274: Can set PTP for ace2 camera

leiwanrobotics, 2025-05-16:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I want to sync my ace2 camera with gps time resource though PTP

### Hardware setup description

Ubuntu22.04
ROS2 humble

### Runtime information

```shell
When I set the ptp through ros service call, I can enable the ptp successfully. Then I Check Pott status through ros service as well, then it shows always waiting for service become available, it seems the camera didn’t get ptp. I also compare the timestamp of image with Unix timestamp, it’s 2 seconds different. But I can set PTP successfully though pylonviewer, it shows offset is around 100ns.
```

### Is your camera operational with the Basler pylon Viewer on your platform?

Yes

FrancoisPicardDTI, 2025-05-20:
Hello @leiwanrobotics 
Just to be sure, so you are trying to synchronize 2 cameras together that's it? Can you describe a bit more your application ?
Can you please post the different ROS commands you are executing ?

nathan1536, 2025-10-31:
Do you solve this problem ?

I also use this command.
-  ros2 service call /my_camera/pylon_ros2_camera_node/enable_ptp std_srvs/srv/SetBool "{data: true}"
-  ros2 service call /my_camera/pylon_ros2_camera_node/get_ptp_status ""
- 
waiting for service to become available...
requester: making request: pylon_ros2_camera_interfaces.srv.GetPtpStatus_Request()

response:
pylon_ros2_camera_interfaces.srv.GetPtpStatus_Response(ptp_status='', ptp_servo_status='', offset_from_master=0, success=False, message='The connected camera does not support this feature')

FrancoisPicardDTI, 2026-11-04:
To summarize:
* Docker Desktop VM: on Linux --network=host attaches to Docker Desktop's internal VM, not your physical NIC, so GigE discovery broadcasts never reach the camera. Use Docker Engine instead.
* Netmask mismatch: if the camera's netmask doesn't match the host interface's, the driver's subnet-broadcast discovery misses the camera (pylon Viewer still works because it also uses the limited broadcast 255.255.255.255). Set the camera netmask to match the interface and verify with tcpdump.

I am clossing the issue. Please reopen if needed.

closed 2026-11-04 by FrancoisPicardDTI