basler/pylon-ros-camera #275: Driver doesn't respect `trigger_timeout`

diablo1281, 2025-05-29:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I am trying to set up a Basler Ace 2 USB camera with ROS2. I want to use an external trigger to sync multiple cameras on a drone and take pictures at a given moment of the flight, dictated by the autopilot. Most important is that the constant rate can't be set in my situation.

Testing with a single camera - it is configured to accept a trigger on Line 1, and it works like a charm.

During the start of the ROS2 driver `pylon_ros2_camera_wrapper`, I need to spam the trigger. If not, I get an error and the thread shuts down, giving output: 

```
[pylon_ros2_camera_wrapper-1] 1748550317.880437163 [basler.pylon.ros2.pylon_ros2_camera_base] [ERROR] Waiting for Trigger signal
[pylon_ros2_camera_wrapper-1] 1748550317.880509997 [basler.pylon.ros2.pylon_ros2_camera_base] [ERROR] PylonROS2Camera not ready because the result of the initial grab is invalid
[pylon_ros2_camera_wrapper-1] 1748550317.880528205 [basler.pylon.ros2.pylon_ros2_camera_node] [ERROR] Error when trying to start grabbing. Shutting down now.
[pylon_ros2_camera_wrapper-1] Impossible to spin
[pylon_ros2_camera_wrapper-1] failed to create guard condition: the given context is not valid, either rcl_init() was not called or rcl_shutdown() was called., at ./src/rcl/guard_condition.c:67
```

I tried to set `trigger_timeout` to some big number, but it doesn't make any difference.

In case I manage to spam trigger to get to launch driver - there is second problem.
If there is any subscriber to the camera topic, it keep to output message:
```
[pylon_ros2_camera_wrapper-1] 1748551306.421796720 [basler.pylon.ros2.pylon_ros2_camera_base] [ERROR] Waiting for Trigger signal
[pylon_ros2_camera_wrapper-1] 1748551306.421867345 [basler.pylon.ros2.pylon_ros2_camera_base] [ERROR] Error: Grab was not successful
[pylon_ros2_camera_wrapper-1] 1748551306.926431561 [basler.pylon.ros2.pylon_ros2_camera_base] [ERROR] Waiting for Trigger signal
[pylon_ros2_camera_wrapper-1] 1748551306.926495915 [basler.pylon.ros2.pylon_ros2_camera_base] [ERROR] Error: Grab was not successful
[pylon_ros2_camera_wrapper-1] 1748551307.430591448 [basler.pylon.ros2.pylon_ros2_camera_base] [ERROR] Waiting for Trigger signal
[pylon_ros2_camera_wrapper-1] 1748551307.430720603 [basler.pylon.ros2.pylon_ros2_camera_base] [ERROR] Error: Grab was not successful
```

I would like to have a camera thread publishing data only then the photo was taken (by trigger) and the driver to initialize properly without the need to make a capture.

### Hardware setup description

- Jetson Orin NX 16GB
- Auvidea JNX45 carrier board
- Bassler ace 2 a2A4508-20ucBAS
- ROS2 humble (running in docker)

### Runtime information

```shell
pylon info: 9.0.3.215 64-Bit
----------------------------------------
pylon Setup info: 8.0.1.16188 
----------------------------------------
pylon Application info: 3.3.0.16267 64-Bit
----------------------------------------
pylon Viewer info: 9.0.2.16267 64-Bit
----------------------------------------
Host system infos:

CPU architecture: arm64
Kernel type: linux
Kernel version: 5.15.148-tegra
OS type: ubuntu
OS version: 22.04
Render Backend: OpenGL
----------------------------------------
Screen(s):

Screen 0 scale factor: 1
Screen 0 resolution: 2560x1440
----------------------------------------
Window(s):

"pylon Viewer 64-Bit" on screen 0
----------------------------------------
Host network infos:

Host network adapter enP8p1s0:
    Name: enP8p1s0
    MAC address: 48:B0:2D:EA:F5:21
    IP address: 10.41.10.1
    Subnet mask: 255.255.255.0

Host network adapter l4tbr0:
    Name: l4tbr0
    MAC address: 3E:C7:E7:79:F3:87

Host network adapter usb0:
    Name: usb0
    MAC address: 9E:45:D6:AB:26:E1

Host network adapter usb1:
    Name: usb1
    MAC address: 9E:45:D6:AB:26:E3

Host network adapter wlP7p1s0:
    Name: wlP7p1s0
    MAC address: F0:20:FF:AD:D5:E0
    IP address: 192.168.1.144
    Subnet mask: 255.255.255.0

Host network adapter docker0:
    Name: docker0
    MAC address: 4E:D0:30:BF:E5:01
    IP address: 172.17.0.1
    Subnet mask: 255.255.0.0
----------------------------------------
Device infos:

USB3 Vision device:
    Model name: a2A4508-20ucBAS
    Manufacturer name: Basler
    Serial number: 40544869
    Device GUID: 267601CBA0FE
    Product ID: 0xBA05
    Vendor ID: 0x2676
    USB version: 3.2.0
    Device path: 2676:ba05:2:4:4

USB3 Vision device:
    Model name: a2A4508-20ucBAS
    Manufacturer name: Basler
    Serial number: 40544868
    Device GUID: 267601CBA0FB
    Product ID: 0xBA05
    Vendor ID: 0x2676
    USB version: 3.2.0
    Device path: 2676:ba05:2:3:5

USB3 Vision device:
    Model name: a2A4508-20ucBAS
    Manufacturer name: Basler
    Serial number: 40202683
    Device GUID: 267601CB202E
    Product ID: 0xBA05
    Vendor ID: 0x2676
    USB version: 3.2.0
    Device path: 2676:ba05:2:2:3
```

### Is your camera operational with the Basler pylon Viewer on your platform?

Yes

FrancoisPicardDTI, 2025-07-08:
Hello @diablo1281 

I need more info about your problem:

- You're saying that it works like a charm with one camera. What is? Can you describe what is working?
- When you set the trigger timeout, how do you do it? Do you get an error message?
- What do you mean by spam the trigger? What do you do exactly? So here, you are not working with one camera anymore, right?
- And you are writing "In case I manage to spam trigger to get to launch driver", so sometimes your "spamming" is working, and sometimes not?

Let's try to go deeper with these points in the first place.

FrancoisPicardDTI, 2026-08-20:
Hi @diablo1281
Did you have the chance to look at the new code version and check if it solves your issue?

FrancoisPicardDTI, 2026-10-29:
Closing the issue for now, reopen if needed.

closed 2026-10-29 by FrancoisPicardDTI