basler/pylon-ros-camera #277: BUG: StartTimeLow for Synchronous Free Run with time-aligned intervals cannot be set

lucleon, 2025-06-03:
### Issue description

When I try to run the `set_sync_free_run_timer_start_time_low` service with 

ros2 service call /cam1/pylon_ros2_camera_node/set_sync_free_run_timer_start_time_low pylon_ros2_camera_interfaces/srv/SetIntegerValue "value: 2881282048"

I get the following error message: `Failed to populate field: The 'value' field must be an integer in [-2147483648, 2147483647]` (I took the 32bit integer from an [example in the pylon documentation](https://docs.baslerweb.com/synchronous-free-run#converting-the-64-bit-timestamp-to-start-time-high-and-start-time-low) and verified that in the PylonViewer StartTimeLow can be set to this value).

However, when I try to pass the signed representation of the binary number to the service call I get another error:
`response:
pylon_ros2_camera_interfaces.srv.SetIntegerValue_Response(success=False, message='Value = -1413685248 must be equal or greater than Min = 0.')`

Am I using the service set_sync_free_run_timer_start_time_low in a wrong way? Or is there a bug in the service implementation?

### Code example about how to reproduce the issue

```
ros2 service call /cam1/pylon_ros2_camera_node/set_sync_free_run_timer_start_time_low pylon_ros2_camera_interfaces/srv/SetIntegerValue "value: 2881282048"
```

### Error message

```shell

```

### Hardware setup description

Ubuntu 22.04

### Runtime information

```shell
ROS2 Humble
```

### Is your camera operational with the Basler pylon Viewer on your platform?

Yes

SMA2016a, 2025-06-04:
the camera parameter is an uint32.  Is it possible to change the variable form int32 to int64

SMA2016a, 2025-07-08:
Thanks for the Update.

closed 2025-08-07 by FrancoisPicardDTI